VersBinarii/bme280-rs #3: I2C address inconsistency

ryankurte, 2019-08-15:
hey thanks for making a neat library!

i notice in the [docs](https://github.com/uber-foo/bme280-rs/blob/a3d9383a79b43845ffc9712a6d01ac3fd78f9127/src/lib.rs#L40) the primary address is listed as `0x77` with secondary as `0x78`, whereas afaik in the datasheet this should be `0x77` and `0x76`, and in the [constants](https://github.com/uber-foo/bme280-rs/blob/a3d9383a79b43845ffc9712a6d01ac3fd78f9127/src/lib.rs#L64) `primary` is set to `0x76` and `secondary` is set to `0x77` which is inconsistent with the documentation.

i was a bit bamboozled me a little until i manually set the address. can open a PR but i’m not sure which way around you’d prefer to fix this?

sbruton, 2019-08-15:
Thanks for catching the inconsistency and for opening an issue!

From the datasheet (page 32):
```
Connecting SDO to GND results in slave address 1110110 (0x76);
connection it to VDDIO results in slave address 1110111 (0x77)...
```

So it looks like, according to the datasheet, the code is correct and the inline documentation is wrong. Since I've had no trouble with this library in practice, I think it's safe to update the docs. I'll open a PR for this in a moment and bump the version on crates.io.

Thanks again!

sbruton, 2019-08-15:
crates.io updated with [version 0.1.2](https://crates.io/crates/bme280), thanks again @ryankurte

PR #4 (2019-08-15): updates incorrect I2C address in documentation, cleans up formatting …
…of compiler flags, fixes #3
commit ec8cc82a (2019-08-15): updates incorrect I2C address in documentation, cleans up formatting of compiler flags, fixes #3
closed 2019-08-15 by sbruton