VersBinarii/bme280-rs #5: Driver panics when trying to make a measurement with BME280

Glaeqen, 2020-04-23:
Hey!
I love your library, looks overall pretty clean. However, I'm struggling to spin it up. I created a simple program (very similar to your example) running on the STM32F407. Everything works fine (temperature and pressure is calculated correctly) with [bmp280_ehal](https://docs.rs/bmp280-ehal/0.0.3/bmp280_ehal/) driver as BME280 is backward compatible (somewhat) to BMP280. It's downside is that it lacks humidity support. 

When trying to use your library, it panics when it tries to make a measurement (`bme280::BME280::measure()`) - after query to device and applying compensations, pressure appears to be equal 0, so condition is not fulfilled and [panic is raised here](https://github.com/uber-foo/bme280-rs/blob/4b0652baea4b2cbe340c72efe7476a921a040600/src/lib.rs#L281)

Sample source code
```rust
#![no_std]
#![no_main]

extern crate panic_halt;
extern crate stm32f4;
use cortex_m_rt::entry;
use stm32f4xx_hal::i2c::I2c;
use stm32f4xx_hal::gpio::GpioExt;
use stm32f4xx_hal::rcc::RccExt;
use stm32f4xx_hal::time::U32Ext;

#[entry]
fn main() -> ! {
    let stm32_specific_peripherals = stm32f4::stm32f407::Peripherals::take().unwrap();
    let cortex_m_peripherals = cortex_m::Peripherals::take().unwrap();
    let clocks = stm32_specific_peripherals.RCC.constrain().cfgr.sysclk(168.mhz()).freeze();

    // I2C
    let gpiob = stm32_specific_peripherals.GPIOB.split();
    let i2c = I2c::i2c1(
        stm32_specific_peripherals.I2C1,
        (gpiob.pb6.into_alternate_af4_open_drain(), gpiob.pb7.into_alternate_af4_open_drain()),
        400.khz(),
        clocks,
    );

    let delay_provider = stm32f4xx_hal::delay::Delay::new(cortex_m_peripherals.SYST, clocks);
    let mut bme280 = bme280::BME280::new_primary(i2c, delay_provider);
    bme280.init().unwrap();

    loop {
        let measurements = bme280.measure().unwrap();
        // .. do stuff
    }
}
```

There's a zipped [Logic Saleae](https://www.saleae.com/downloads/) I2C transmission capture dump provided in attachment that might contain some helpful info regarding issue: [i2c_communication.zip](https://github.com/uber-foo/bme280-rs/files/4525877/i2c_communication.zip)

If you were to just point me to the right direction, I'd be very grateful. Thank you for your help!

sbruton, 2020-04-25:
Thanks for the spectacular detail here. Nothing jumps out at me with your sample code above. The closest thing to your F4 that I've used this crate on is an STM32F042K6. I do have an STM32F4 dev board here, so I'll can see if I can reproduce.

I'm unfortunately in the middle of a time-critical project and may not be able to look at this for a couple of weeks.

Glaeqen, 2020-04-29:
Ok, thanks for the response :)

Glaeqen, 2020-06-11:
https://github.com/stm32-rs/stm32f4xx-hal/pull/169

Problem was within HAL implementation for STM32F4XX. **STOP condition was not generated** for i2c write calls via `embedded_hal::blocked::i2c::Write::write()`. Fix is already on master, we can expect release soon. Closing.

closed 2020-06-11 by Glaeqen