VersBinarii/bme280-rs #7: Allow sharing the I2C bus.

mgottschlag, 2020-05-02:
The driver should allow sharing the bus with other devices. Other drivers provide one of two mechanisms for this:

1. Allowing the user to "uninitialize" the device again and regain access to the underlying objects. I have implemented this approach at https://github.com/mgottschlag/bme280-rs/tree/destroy (on top of on #6). This approach might have advantages in certain scenarios if it allows better power management (it both allows the user to reinitialize the device, e.g., after temporary power supply deactivation, and allows the HAL to reinitialize the bus when required by low-power modes).

2. Passing the bus object as a mutable reference to all functions requiring it. This approach is used, for example, by the (epd_waveshare)[https://github.com/caemor/epd-waveshare] crate. This approach is vastly superior in that it does not require reinitialization of the device over and over again.

Maybe both should be provided, as each has its potential use cases.

ryankurte, 2020-07-13:
you may be interested in [shared-bus](https://crates.io/crates/shared-bus), which is the common approach for splitting `embedded-hal` peripherals for consumption in multiple drivers without need for either of the listed approaches.

VersBinarii, 2022-04-01:
@ryankurte suggestion is the way to go. I'll close this issue for now.

closed 2022-04-01 by VersBinarii