VersBinarii/bme280-rs #11: RTIC example?

arrowcircle, 2020-09-18:
Hi!
I am struggling to use this driver with RTIC because of delay provider.
What is the best way to use this library with RTIC?

nebelgrau77, 2020-12-17:
@arrowcircle You can get the delay in RTIC from core:

```rust
 #[init]
    fn init(cx: init::Context) -> init::LateResources {
        let dp = cx.device;
        let core = cx.core;

        //clocks and other peripherals defined here

        let mut delay = Delay::new(core.SYST, clocks);
```

Here's an example with a different I2C device:

https://github.com/jamwaffles/ssd1306/blob/master/examples/rtic_dvd.rs

arrowcircle, 2020-12-17:
@nebelgrau77 I already tried this thing, but it does not work.

```
no field `SYST` on type `rtic::Peripherals`
```

nebelgrau77, 2020-12-17:
OK, here's my repo that works: 

https://github.com/nebelgrau77/sensor_data_forwarder_Rust

You need to get SYST from cx.core, not cx.device.

arrowcircle, 2020-12-17:
@nebelgrau77 I use bluepill and it looks like there are no SYST in cx.core of F1.

nebelgrau77, 2020-12-17:
I see... hmmm, try asking in RTIC channel here: https://app.element.io/#/room/#rtic:matrix.org - I didn't realize it might be a board/MCU specific issue.

commit 903b66ec (2022-04-01): Add RTIC example resolves #11
closed 2022-04-01 by VersBinarii