VersBinarii/bme280-rs #15: BME280 not working on ESP32

sehraf, 2021-09-14:
I'm trying to get this running on an ESP32 (with ESP-IDF).
I'm able to initialize the device but any measurement results in `I2c(EspError(-1))` (while `-1` is `ESP_FAIL`).
I was able to fix the error above by fixing what looks like a typo (or are *two* 0xF4 intentional?):
```diff
diff --git a/src/lib.rs b/src/lib.rs
index 6b142e1..8aa4779 100644
--- a/src/lib.rs
+++ b/src/lib.rs
@@ -72,7 +72,7 @@ use serde::Serialize;
 const BME280_I2C_ADDR_PRIMARY: u8 = 0x76;
 const BME280_I2C_ADDR_SECONDARY: u8 = 0x77;
 
-const BME280_PWR_CTRL_ADDR: u8 = 0xF4;
+const BME280_PWR_CTRL_ADDR: u8 = 0xF3;
 const BME280_CTRL_HUM_ADDR: u8 = 0xF2;
 const BME280_CTRL_MEAS_ADDR: u8 = 0xF4;
 const BME280_CONFIG_ADDR: u8 = 0xF5;
```

With that fix in place i can get measurements (as many as i want) until i wait for a bit. After a break of some seconds every measurement is the `I2c(EspError(-1))` again.
A logic analizer shows that the ESP is trying to talk to the BME280 but the transfer gets NACKed:
[image]
*(That's all, there is nothing following on the right side)*

I've also notices that the measured values do not change at all and also look (completely) wrong (60_000 hPa ?!). 
```
measured 21.517174°C
measured 59912.863hPa
measured 63.775578%
```

VersBinarii, 2022-04-01:
The 2 0xF4 are in fact intentional. They are used for accessing different sections within that register.

Did you try to affect the environment (i.e. exhaling on the sensor should affect the humidity and temperature) around the sensor when reading the values from it?

sehraf, 2022-04-01:
Whatever was the problem, it solved by itself. Now everything works (could maybe have been a fluke in the ESP-IDF code :man_shrugging: )

closed 2022-04-01 by sehraf