VersBinarii/bme280-rs #19: Ability to configure oversampling and IIR filter

mrd0ll4r, 2022-05-22:
Hey,

looking through the source, the `configure` function sets a bunch of options, notably the oversampling and IIR filter settings.
For our application, it'd be nice to change those, especially the IIR filter, since we're using this for something like weather monitoring, with a very low sample rate.
Any chance you could add a parameter for configuration values to `init`, maybe?

Best,
Leo

VersBinarii, 2022-05-25:
Its definitely something that can be added. 
I dont have a whole lot of time lately so PR's are welcome.
Otherwise i'll try to look at this over next few weeks.

VersBinarii, 2022-06-05:
Closed with https://github.com/VersBinarii/bme280-rs/pull/20

closed 2022-06-05 by VersBinarii