VersBinarii/bme280-rs #21: Build fails for ESP32

enelson1001, 2022-06-01:
Trying to build (use) bme280-rs with my project.  The build fails with the following  error.
```
error[E0432]: unresolved import `embedded_hal::spi::blocking::Transfer`
 --> /home/ed/.cargo/registry/src/github.com-1ecc6299db9ec823/bme280-0.4.1/src/spi.rs:5:5
  |
5 | use embedded_hal::spi::blocking::Transfer;
  |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `Transfer` in `spi::blocking`

For more information about this error, try `rustc --explain E0432`.
error: could not compile `bme280` due to previous error
```

When I look at my cargo.lock file I see the following.
```
[[package]]
name = "bme280"
version = "0.4.1"
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "afa0ec59ce7903e8f6678855e70140b78f2d7a047d5afadf7b14266b020799ae"
dependencies = [
 "embedded-hal 1.0.0-alpha.8",
]
```

1. It looks like embedded-hal 1.0.0-aplha.8 is being used.
2. When I look at embedded-hal 1.0.0-alpha.8 for spi blocking there is no trait Transfer.  In embedded-hal 1.0.0-alpha.7 there is a Transfer trait.  I may be looking at this wrong since I am new to rust.
3. In bme280-rs git hub repository the cargo.toml list the dependency for embedded-hal = 1.0.0-alpha.7.
4. I am new to rust.  So how do I make my cargo.toml make bme280-rs use embedded-hal = 1.0.0-alpha.7 and not embedded-hal = 1.0.0-alpha.8 if the problem really is because of version 1.0.0-alpha.8 ?

For my project I am using I2c and not spi for the bme280 device.

VersBinarii, 2022-06-04:
Hello,
That was due to the new alpha version being released.
I locked it down to the alpha.7 version in Cargo.toml.
Please delete Cargo.lock file in your project and try latest version: https://crates.io/crates/bme280

enelson1001, 2022-06-05:
I had another library that forced embedded-hal to be 1.0.0-alpha.8  so I ended up updating spi.rs to use alpha.8
Would you like me to submit a pull request?  It could probably use some more updating - not sure if SPIError is needed anymore?

Now I am trying to figure out how to use shared_bus for i2c since I have 2 - i2c devices that share the same bus and eventually want to use threads.

VersBinarii, 2022-06-05:
Well. for the bme280 crate it would suffice for you to just upgrade to version 0.4.4 ;)

closed 2022-06-05 by VersBinarii

VersBinarii, 2022-06-06:
That error references ttgo-bme280. It is not this crate. Are you sure you're not mixing up crates in your project?

enelson1001, 2022-06-06:
ttgo-bme280 is my project.
my project Cargo.toml
```
[package]
name = "ttgo-bme280"
version = "0.1.0"
authors = ["enelson1001 <ednelson5080@gmail>"]
edition = "2018"
resolver = "2"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
pio = ["esp-idf-sys/pio"]

[dependencies]
esp-idf-sys = { version = "0.31.5", features = ["binstart"] }
esp-idf-hal = "0.37.4"
esp-idf-svc = "0.41.3"
embedded-hal = "0.2.7"

anyhow = {version = "1", features = ["backtrace"]}
log = "0.4"

embedded-graphics = "0.7.1"
display-interface = "0.4"
ssd1306 = "0.7"

bme280 = { version = "0.4.4"}

shared-bus = {git = "https://github.com/Rahix/shared-bus.git", version = "0.2.3", branch = "i2c-eh-alpha", features = ["eh-alpha"]}

[build-dependencies]
embuild = "0.29"
anyhow = "1"
```
If I use my patched version of 0.4.1 of bme280-rs I do not receive this error.
I also have added shared-bus which they have updated to use alpha.8.  When I use my patched version of bme280-rs everything compiles and runs.

Not sure what I need to do, to get bme280-rs version 0.4.4 to compile without the above error.  Any suggestions welcomed.

VersBinarii, 2022-06-07:
OK, i see now.
I'm unsure what might be the issue in that case. I cant advise you much beyond deleteing the .lock file and cargo clean.
Do you have this project on github somewhere? I could have a quick look at it to see what might be the issue.

enelson1001, 2022-06-07:
I am still in the process of cleaning up the code.  I will let you known when I have put on github.

I am new to rust but what I take from the error message is the following.
```
esp-idf-hal version 0.37.4 requires embedded-hal '=1.0.0-alpha.8'
bme280-rs requires embedded-hal '=1.0.0-alpha.7'

The rust compiler must chose one or the other but not both.
```

eliliam, 2023-07-24:
I'm having a similar issue, and am getting the following error:
```
error: failed to select a version for `embedded-hal`.
    ... required by package `esp-idf-hal v0.41.1`
    ... which satisfies dependency `esp-idf-hal = "^0.41"` of package `esp32-c3-playground v0.1.0 (/home/eli/Projects/esp_rust/esp32-c3-playground)`
versions that meet the requirements `=1.0.0-alpha.10` are: 1.0.0-alpha.10

all possible versions conflict with previously selected packages.

  previously selected package `embedded-hal v1.0.0-alpha.7`
    ... which satisfies dependency `embedded-hal = "=1.0.0-alpha.7"` of package `bme280 v0.4.4`
    ... which satisfies dependency `bme280 = "^0.4.4"` of package `esp32-c3-playground v0.1.0 (/home/eli/Projects/esp_rust/esp32-c3-playground)`

failed to select a version for `embedded-hal` which could resolve this conflict
```

Here is my Cargo.toml, which was generated via `cargo generate esp-rs/esp-idf-template cargo`:
```
[package]
name = "esp32-c3-playground"
version = "0.1.0"
authors = ["Eli Smith <eli@eliliam.com>"]
edition = "2021"
resolver = "2"
rust-version = "1.66"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]

default = ["std", "hal", "esp-idf-sys/native"]

pio = ["esp-idf-sys/pio"]
all = ["std", "nightly", "experimental", "embassy"]
hal = ["esp-idf-hal", "embedded-svc", "esp-idf-svc"]
std = [
	"alloc",
	"esp-idf-sys/std",
	"esp-idf-sys/binstart",
	"embedded-svc?/std",
	"esp-idf-hal?/std",
	"esp-idf-svc?/std",
]
alloc = ["embedded-svc?/alloc", "esp-idf-hal?/alloc", "esp-idf-svc?/alloc"]
nightly = [
	"embedded-svc?/nightly",
	"esp-idf-svc?/nightly",
] # Future: "esp-idf-hal?/nightly"
experimental = ["embedded-svc?/experimental", "esp-idf-svc?/experimental"]
embassy = [
	"esp-idf-hal?/embassy-sync",
	"esp-idf-hal?/critical-section",
	"esp-idf-hal?/edge-executor",
	"esp-idf-svc?/embassy-time-driver",
	"esp-idf-svc?/embassy-time-isr-queue",
]

[dependencies]
log = { version = "0.4.17", default-features = false }
esp-idf-sys = { version = "0.33", default-features = false }
esp-idf-hal = { version = "0.41", optional = true, default-features = false }
esp-idf-svc = { version = "0.46", optional = true, default-features = false }
embedded-svc = { version = "0.25", optional = true, default-features = false }
bme280 = "0.4.4"

[patch.crates-io]
esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc" }

[build-dependencies]
embuild = "0.31.2"
```

The Cargo.toml file is entirely untouched from what was generated aside from the bme280 dependency.

@enelson1001, were you able to find a solution for your similar issue?

eliliam, 2023-07-24:
It appears that while using 0.4.1 does resolve the dependency issues, it also seems to be introducing another bug. The following is what is output with version 0.4.1:
```
   Compiling bme280 v0.4.1
error[E0432]: unresolved import `embedded_hal::delay::blocking`
 --> /home/eli/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bme280-0.4.1/src/i2c.rs:3:26
  |
3 | use embedded_hal::delay::blocking::DelayUs;
  |                          ^^^^^^^^ could not find `blocking` in `delay`

error[E0432]: unresolved import `embedded_hal::i2c::blocking`
 --> /home/eli/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bme280-0.4.1/src/i2c.rs:4:25
  |
4 | use embedded_hal::i2c::{blocking::I2c, ErrorType};
  |                         ^^^^^^^^ could not find `blocking` in `i2c`

error[E0432]: unresolved import `embedded_hal::delay::blocking`
 --> /home/eli/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bme280-0.4.1/src/spi.rs:3:26
  |
3 | use embedded_hal::delay::blocking::DelayUs;
  |                          ^^^^^^^^ could not find `blocking` in `delay`

error[E0432]: unresolved import `embedded_hal::digital::blocking`
 --> /home/eli/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bme280-0.4.1/src/spi.rs:4:28
  |
4 | use embedded_hal::digital::blocking::OutputPin;
  |                            ^^^^^^^^ could not find `blocking` in `digital`

error[E0432]: unresolved import `embedded_hal::spi::blocking`
 --> /home/eli/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bme280-0.4.1/src/spi.rs:5:24
  |
5 | use embedded_hal::spi::blocking::Transfer;
  |                        ^^^^^^^^ could not find `blocking` in `spi`

error[E0432]: unresolved import `embedded_hal::delay::blocking`
  --> /home/eli/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bme280-0.4.1/src/lib.rs:69:26
   |
69 | use embedded_hal::delay::blocking::DelayUs;
   |                          ^^^^^^^^ could not find `blocking` in `delay`

For more information about this error, try `rustc --explain E0432`.
error: could not compile `bme280` (lib) due to 6 previous errors
warning: build failed, waiting for other jobs to finish...
```

And a similar error with 0.3.0 is happening as well:
```
error[E0277]: the trait bound `&esp_idf_hal::i2c::I2cDriver<'_>: embedded_hal::blocking::i2c::Read` is not satisfied
  --> src/main.rs:28:35
   |
28 |     let bme280 = BME280::new_primary(&i2c);
   |                  ------------------- ^^^^ the trait `embedded_hal::blocking::i2c::Read` is not implemented for `&esp_idf_hal::i2c::I2cDriver<'_>`
   |                  |
   |                  required by a bound introduced by this call
   |
note: required by a bound in `bme280::i2c::BME280::<I2C, D>::new_primary`
  --> /home/eli/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bme280-0.3.0/src/i2c.rs:22:10
   |
22 |     I2C: Read<Error = E> + Write<Error = E> + WriteRead<Error = E>,
   |          ^^^^^^^^^^^^^^^ required by this bound in `BME280::<I2C, D>::new_primary`
...
26 |     pub fn new_primary(i2c: I2C, delay: D) -> Self {
   |            ----------- required by a bound in this associated function
help: consider removing the leading `&`-reference
   |
28 -     let bme280 = BME280::new_primary(&i2c);
28 +     let bme280 = BME280::new_primary(i2c);
   |

error[E0277]: the trait bound `&esp_idf_hal::i2c::I2cDriver<'_>: embedded_hal::blocking::i2c::Write` is not satisfied
  --> src/main.rs:28:35
[... 36 lines cut ...]
help: consider removing the leading `&`-reference
   |
28 -     let bme280 = BME280::new_primary(&i2c);
28 +     let bme280 = BME280::new_primary(i2c);
   |

error[E0061]: this function takes 2 arguments but 1 argument was supplied
  --> src/main.rs:28:15
   |
28 |     let bme280 = BME280::new_primary(&i2c);
   |                  ^^^^^^^^^^^^^^^^^^^------ an argument is missing
   |
note: associated function defined here
  --> /home/eli/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bme280-0.3.0/src/i2c.rs:26:12
   |
26 |     pub fn new_primary(i2c: I2C, delay: D) -> Self {
   |            ^^^^^^^^^^^
help: provide the argument
   |
28 |     let bme280 = BME280::new_primary(&i2c, /* delay */);
   |                                     ~~~~~~~~~~~~~~~~~~~

Some errors have detailed explanations: E0061, E0277.
For more information about an error, try `rustc --explain E0061`.
```

eliliam, 2023-07-24:
While using version 0.4.4 and recreacting the Cargo.lock file as well as running `cargo clean`, I am having issues building the library. My Cargo.lock now contains the following under the bme280 dependency:
```
[[package]]
name = "bme280"
version = "0.4.1"
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "afa0ec59ce7903e8f6678855e70140b78f2d7a047d5afadf7b14266b020799ae"
dependencies = [
 "embedded-hal 1.0.0-alpha.10",
]
```

And here is the error I am getting while building:
```
error[E0432]: unresolved import `embedded_hal::delay::blocking`
 --> /home/eli/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bme280-0.4.1/src/i2c.rs:3:26
  |
3 | use embedded_hal::delay::blocking::DelayUs;
  |                          ^^^^^^^^ could not find `blocking` in `delay`

error[E0432]: unresolved import `embedded_hal::i2c::blocking`
 --> /home/eli/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bme280-0.4.1/src/i2c.rs:4:25
  |
4 | use embedded_hal::i2c::{blocking::I2c, ErrorType};
  |                         ^^^^^^^^ could not find `blocking` in `i2c`

error[E0432]: unresolved import `embedded_hal::delay::blocking`
 --> /home/eli/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bme280-0.4.1/src/spi.rs:3:26
  |
3 | use embedded_hal::delay::blocking::DelayUs;
  |                          ^^^^^^^^ could not find `blocking` in `delay`

error[E0432]: unresolved import `embedded_hal::digital::blocking`
 --> /home/eli/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bme280-0.4.1/src/spi.rs:4:28
  |
4 | use embedded_hal::digital::blocking::OutputPin;
  |                            ^^^^^^^^ could not find `blocking` in `digital`

error[E0432]: unresolved import `embedded_hal::spi::blocking`
 --> /home/eli/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bme280-0.4.1/src/spi.rs:5:24
  |
5 | use embedded_hal::spi::blocking::Transfer;
  |                        ^^^^^^^^ could not find `blocking` in `spi`

error[E0432]: unresolved import `embedded_hal::delay::blocking`
  --> /home/eli/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bme280-0.4.1/src/lib.rs:69:26
   |
69 | use embedded_hal::delay::blocking::DelayUs;
   |                          ^^^^^^^^ could not find `blocking` in `delay`

For more information about this error, try `rustc --explain E0432`.
error: could not compile `bme280` (lib) due to 6 previous errors
warning: build failed, waiting for other jobs to finish...
```