VersBinarii/bme280-rs #25: SPI communication doesn't work

wiktorwieclaw, 2023-01-08:
The following code is not right, as it would require the data to be prepended with a dummy byte
```rust
impl<SPI, CS> SPIInterface<SPI, CS>
where
    SPI: Transfer<u8>,
    CS: OutputPin,
{
    fn read_any_register(
        &mut self,
        register: u8,
        data: &mut [u8],
    ) -> Result<(), Error<SPIError<SPI::Error, CS::Error>>> {
        self.cs
            .set_low()
            .map_err(|e| Error::Bus(SPIError::Pin(e)))?;
        self.spi
            .transfer(data, &[register])
            .map_err(|e| Error::Bus(SPIError::SPI(e)))?;
        self.cs
            .set_high()
            .map_err(|e| Error::Bus(SPIError::Pin(e)))?;
        Ok(())
    }
}
```
Spitting the transfer would do the job:
```rust
bus.write(&[register])?;
bus.read(data)
```

VersBinarii, 2023-01-16:
Fixed with https://github.com/VersBinarii/bme280-rs/pull/23

wiktorwieclaw, 2023-01-16:
@VersBinarii I believe that #23 doens't fix the issue. The actual fix is this:

From:
```rust
self.spi
    .transfer(data, &[register])
    .map_err(|e| Error::Bus(SPIError::SPI(e)))?;
```
To:
```rust
self.device
    .transaction(|bus| {
        bus.write(&[register])?;
        bus.read(data)
     })
```

Please reopen the issue, I'll prepare a new, updated PR that resolves conflicts.

VersBinarii, 2023-01-16:
Have you tested it though?
I dont have the hardware on my desk to confirm it.
I'll get some by the end of the week.

wiktorwieclaw, 2023-01-16:
@VersBinarii Yup. I'll test it again on Wednesday just to be sure.

For now I could try to explain why the current approach doesn't work.

Look at impl of `transfer` from [stm32f4xx_hal](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/spi/hal_1.rs#L88):
```rust
fn transfer(&mut self, buff: &mut [W], data: &[W]) -> Result<(), Self::Error> {
    assert_eq!(data.len(), buff.len());

    for (d, b) in data.iter().cloned().zip(buff.iter_mut()) {
         nb::block!(<Self as FullDuplex<W>>::write(self, d))?;
         *b = nb::block!(<Self as FullDuplex<W>>::read(self))?;
     }

     Ok(())
}
```
Let's say we want to read 3 bytes from address `0xb7`. For each byte we want to read, we need to provide a dummy byte to write. So, something like this wouldn't work:
```rust
let write = [0xb7];
let mut read = [0x00; 3];
spi.transfer(&write, &mut read)
```
If we really want to use `transfer` method, we shall do something this:
```rust
let write = [0xb7, 0x00, 0x00, 0x00];
let mut read = [0x00, 0x00, 0x00, 0x00];
spi.transfer(&write, &mut read);

let response = &read[1..];
```

[HAL_SPI_TransmitReceive](https://stackoverflow.com/a/67936545) from the C HAL works exactly same.

Note that I didn't use `transfer` in my PR to avoid slicing and copying the response from the read buffer.

VersBinarii, 2023-01-17:
I see, you're correct.
Feel free to make a PR then if you wish 😁 If not i'll look at fixing this later this week.

wiktorwieclaw, 2023-01-20:
FYI, I connected everything today, but I'll finish the PR tomorrow :P

PR #43 (2025-04-06): Fix SPI communication
This fixes #25.
    
Any try to call the `init` function will fail with an `UnsupportedChip` error without this (using the default `sync` feature).
    
Previous attempts were made to fix this (#28 and #38) but these where made before `embedded-hal` `1.0.0` and currently have conflicts  preventing their merge (and can't directly be used in a project using `embedded-hal` `1.0.0`
    
closes #28
commit bf5ea17f (2025-04-06): Fix SPI communication (#43)

* add intellij/rustrover files to gitignore

* Fix SPI comunication

This fixes #25.

Any try to call the `init` function will fail with an `UnsupportedChip`
error without this (using the defautl `sync` feature).

Previous attempts were made to fix this (#28 and #38) but these where
made before `embedded-hal` `1.0.0` and currently have conflicts
preventing their merge (and can't directly be used in a project using
`embedded-hal` `1.0.0`

closes #28
closed 2025-04-06 by VersBinarii