VersBinarii/bme280-rs #31: Is a pinned embedded-hal necessary?

oddlama, 2023-08-23:
Is there any reason why embedded-hal is pinned to a specific version? Usually drivers seem to require only a minimum version. This unfortunately causes this crate to break each time e.g. embassy bumps it's embedded hal dependency.

avsaase, 2023-12-10:
FYI, `embedded-hal` and `embedded-hal-async` are scheduled to release 1.0 on December 28th when Rust 1.75 releases. I think `embassy` also plans to do a release that day so from that point forward hopefully we will have less frequent breakage due to incompatible versions.

VersBinarii, 2024-01-22:
Unfortunately it was needed at the time to work around problem with embedded hal released at the time. It should not be needed with recent releases though.

closed 2024-01-22 by VersBinarii