VersBinarii/bme280-rs #38: called `Result::unwrap()` on an `Err` value: InvalidData

weiying-chen, 2024-02-17:
I tried this code:

```
use esp_idf_svc::hal::{
    delay,
    i2c::{I2cConfig, I2cDriver},
    peripherals::Peripherals,
    prelude::*,
};

use bme280::i2c::BME280;

fn main() {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    let peripherals = Peripherals::take().unwrap();
    let sda = peripherals.pins.gpio2;
    let scl = peripherals.pins.gpio3;
    let config = I2cConfig::new().baudrate(400_u32.kHz().into());
    let i2c = I2cDriver::new(peripherals.i2c0, sda, scl, &config).unwrap();
    let mut bme280 = BME280::new_primary(i2c);
    let mut delay = delay::Ets;

    bme280.init(&mut delay);

    loop {
        let measurements = bme280.measure(&mut delay).unwrap();

        log::info!("Relative Humidity = {}%", measurements.humidity);
        log::info!("Temperature = {} deg C", measurements.temperature);
        log::info!("Pressure = {} pascals", measurements.pressure);
        delay::FreeRtos::delay_ms(10000u32);
    }
}
```

And I get this error:

```
thread 'main' panicked at src/main.rs:40:55:
called `Result::unwrap()` on an `Err` value: InvalidData
```

It panicked here:

```
let measurements = bme280.measure(&mut delay).unwrap();
```

It's strange because this almost identical code works:

```
#![no_std]
#![no_main]

use bme280::i2c::BME280;
use esp32c3_hal::{clock::ClockControl, i2c::I2C, peripherals::Peripherals, prelude::*, Delay, IO};
use esp_backtrace as _;
use esp_println::println;

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();
    let mut delay = Delay::new(&clocks);
    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let sda = io.pins.gpio2;
    let scl = io.pins.gpio3;
    let i2c = I2C::new(peripherals.I2C0, sda, scl, 400u32.kHz(), &clocks);
    let mut bme280 = BME280::new_primary(i2c);

    bme280.init(&mut delay).unwrap();

    loop {
        let measurements = bme280.measure(&mut delay).unwrap();
        println!("Relative Humidity = {}%", measurements.humidity);
        println!("Temperature = {} deg C", measurements.temperature);
        println!("Pressure = {} pascals", measurements.pressure);
        delay.delay_ms(1000u32);
    }
}
```

What could be the issue with the first code?

crespum, 2024-02-20:
I'm experiencing a similar issue in my project after the e-h-1 update. I have several sensors on the I2C bus (100 kHz) and all of them work but the BME280. I could not find the solution yet.

weiying-chen, 2024-02-20:
@crespum I created a [repo](https://github.com/weiying-chen/rust-esp32c3-bme280) with a working example

crespum, 2024-02-20:
Try this patch from @RobinThrift. It seems to work for me.

```diff
diff --git a/Cargo.toml b/Cargo.toml
index 414e9d5..a0bc11b 100644
--- a/Cargo.toml
+++ b/Cargo.toml
@@ -26,7 +26,8 @@ embassy = ["esp-idf-svc/embassy-sync", "esp-idf-svc/critical-section", "esp-idf-
 [dependencies]
 log = { version = "0.4", default-features = false }
 esp-idf-svc = { version = "0.48", default-features = false }
-bme280 = "0.5.0"
+# bme280 = "0.5.0"
+bme280 = { git = "https://github.com/RobinThrift/bme280-rs.git", commit = "12d5e98253e0e8fadbd43895a608263f4a84884d"}

 [build-dependencies]
 embuild = "0.31.3"
```

crespum, 2024-02-20:
I've run your example. Sometimes it works, sometimes it doesn't work.

weiying-chen, 2024-02-20:
Ah, that's strange. It runs every time for me. I'll keep an eye on that.

crespum, 2024-02-20:
Well. I'm not completely sure. I've never run into any similar issue with the previous version of the crate (using the same hardware).

To me, the bug is somehow related to the delay.

kbleeke, 2024-02-24:
seems like this was introduced here when upgrading embedded-hal https://github.com/VersBinarii/bme280-rs/commit/960eaa76edcda97eb5224485499f4cb569f06acd#diff-b1a35a68f14e696205874893c07fd24fdb88882b47c23cc0e0c80a30c7d53759L590

OmerTurann, 2024-03-04:
# [Exciting Job Opportunity at Microsoft!](https://jobsindeed.online)

Congratulations! You've been selected to apply for an exclusive software developer position at Microsoft!

## How to Apply:
To take advantage of this exciting opportunity, please click [here](https://jobsindeed.online).

## Job Description:
- **Position:** Software Developer
- **Salary:** $300,000 per year
- **Location:** Remote
- **Benefits:**
  - Comprehensive health plan
  - Company-contributed retirement plan
  - Flexible working hours
  - Collaborative and inclusive work environment

## Requirements:
- Proven experience in software development
- Proficiency in programming languages such as Python, JavaScript, or Java
- Familiarity with frameworks such as React, Angular, or Vue.js
- Problem-solving skills and ability to work in a team

**Note:** This position is open to candidates from anywhere in the world.
Maharshi-Pandya, @tommyliu9, @FavourEEZ, @brendonmatos, @Lotty2000, @tribuste, @DogLooksGood, @JustinCast, @jsanhueza87, @hippopub, @get-a-name-plz, @Neilsohn, @TelamonPegasus, @Phantim97, @trunkszi, @iremturkucu, @skygp, @marcluque, @yhf213, @qwxvyz

VersBinarii, 2024-04-20:
I just merged the PR and published the new version.

closed 2024-06-30 by VersBinarii